Pi4J/pi4j-v1 #545: Can I use this library for other boards?

Dhina17, 2021-09-29:
I want to use this library for dragonboard410c for i2c device handling. I have seen the code. It seems generic. But still i always get UnsupportedBusNumberException. 

Just want to make sure if it can be used for other boards too. 
This is more than enough. 

Thank you.

FDelporte, 2021-09-30:
Theoretically yes, but not implemented. If you look at version 2 of the library, a system of platforms and providers is included which would allow adding support for other boards than Raspberry Pi.

Raspberry Pi: https://github.com/Pi4J/pi4j-v2/tree/develop/plugins/pi4j-plugin-raspberrypi/src/main/java/com/pi4j/plugin/raspberrypi

Start of documentation is available on: https://pi4j.com/documentation/platforms/

Dhina17, 2021-10-02:
Got it. Thank you for your time.

closed 2021-10-02 by Dhina17