Pi4J/pi4j-v1 #550: PCIe x1 connector

alan543, 2022-02-10:
請問圈起來的P & N pin 是原廠線路否有畫錯筆誤
[image]

[image]

正確
[image]

FDelporte, 2022-02-10:
Sorry, not something I can answer as the Pi4J project doesn't provide PCIE access.

closed 2022-02-10 by FDelporte

alan543, 2022-02-11:
> 抱歉，我無法回答，因為 Pi4J 項目不提供 PCIE 訪問。

sorry, I don't understand what you mean, I want to express whether the original circuit has drawn P(+)&N(-) backwards,THX

alan543, 2022-02-11:
> Sorry, not something I can answer as the Pi4J project doesn't provide PCIE access.

> Sorry, not something I can answer as the Pi4J project doesn't provide PCIE access.

sorry, I don't understand what you mean, I want to express whether the original circuit has drawn P(+)&N(-) backwards,THX

alan543, 2022-02-11:
> Sorry, not something I can answer as the Pi4J project doesn't provide PCIE access.

sorry, I don't understand what you mean, I want to express whether the original circuit has drawn P(+)&N(-) backwards,THX

FDelporte, 2022-02-11:
I think this is a question that will get a better answer on https://raspberrypi.stackexchange.com/